RichardMidnight/pi-safe #1: SD Card Copier

slashtechno, 2021-02-13:
First off I would like to say that I really like the idea of Pi-Safe. An improvement that I thought of would be an SD card copier. This would copy the internal storage to an external storage device.

Best Regards, SlashTechno

RichardMidnight, 2021-02-13:
SlashTechno, thanks for your suggestion.  The Raspberry Pi comes with a good tool for doing that: "SD Card Copier".  You can start it from the menu or typing "piclone" in a terminal.  
Pi-Safe can copy the internal storage by setting "allow root backup" to 'yes'.  It will store it in a compressed image file (.zip) which can be restored to any sd or usb device it will fit on.  It will resize the partition to fill the device upon restore.  PiSafe won't go directly to a device like "SD Card Copier" does.  I am not sure how I would do that, but I will think about it.  Thanks.

Best regards, RichardMidnight

slashtechno, 2021-02-13:
Okay thanks.

closed 2021-02-13 by slashtechno