RichardMidnight/pi-safe #2: Zpaq support

Sanesecurity, 2021-02-15:
Could Zpaq support be added... 

http://mattmahoney.net/dc/zpaq.html

As its got journaling built in, it may make it a good choice for snapshots and saving a lot of space. 

Thanks

RichardMidnight, 2021-02-17:
Sanesecurity thanks for the suggestion.  I need to polish the current feature set first.  Then I can look into this.

Peace

RichardMidnight, 2021-03-12:
Thanks for the heads up on Zpaq.
I just took a look at Zpaq.  It looks like a good compression tool for incremental backups.  Since PiSafe does a full image backup, the incremental backup functions of Zpaq are not helpful.  I ran a quick test and Zpaq did not outperform zip, or gz.  If you do want to compress with Zpaq, simply use PiSafe to create a .img file.  Then compress that with Zpaq, or any tool you want.
Peace

closed 2021-03-12 by RichardMidnight