RichardMidnight/pi-safe #3: pi-safe install issue in ubuntu server 20.04 LTS 2

arsirantala, 2021-02-17:
Hi,

I did not have lua installed in the ubuntu server initially. I installed it with following command:

`sudo apt-get install liblua5.1-0-dev liblua50-dev liblualib50-dev`

Then I attempted again to install pi-safe and got following errors:

`bash pisafe                                                                                                                                                            ─╯
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read default_path: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read default_extension: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read compression_level: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read allow_root_backup: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read debug_mode: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read sound: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read shrink_fs: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read safeties: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read default_path: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read default_extension: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read compression_level: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read allow_root_backup: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read debug_mode: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read sound: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read shrink_fs: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read safeties: No such file or directory
mkdir: cannot create directory ‘’: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read default_path: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read default_extension: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read compression_level: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read allow_root_backup: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read debug_mode: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read sound: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read shrink_fs: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read safeties: No such file or directory
mkdir: cannot create directory ‘’: No such file or directory
pisafe: line 49: [: =: unary operator expected
pisafe: line 52: [: =: unary operator expected
pisafe: line 55: [: =: unary operator expected
pisafe: line 58: [: !=: unary operator expected
pisafe: line 61: [: =: unary operator expected
pisafe: line 64: [: =: unary operator expected
pisafe: line 67: [: =: unary operator expected
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read default_path: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read default_extension: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read compression_level: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read allow_root_backup: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read debug_mode: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read sound: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read shrink_fs: No such file or directory
lua: stdin:1: attempt to index global `arg' (a nil value)
stack traceback:
	stdin:1: in main chunk
	[C]: ?
lua: cannot read safeties: No such file or directory
mkdir: cannot create directory ‘’: No such file or directory
Install needed tools:  PiShrink, zip,  and xz-tools?`

Any ideas what lua version I need to install ubuntu server? I presume ubuntu server is supported?

arsirantala, 2021-02-17:
Lua version is:
Lua 5.0.3  Copyright (C) 1994-2006 Tecgraf, PUC-Rio

arsirantala, 2021-02-17:
Found a way to get the pisafe to work - had to install newer version of lua:

sudo apt install lua5.3

Case closed :)

closed 2021-02-17 by arsirantala

RichardMidnight, 2021-03-12:
New version 0.9.18 should address that.
Thanks for letting me know.