RichardMidnight/pi-safe #6: can't find SD card

jfabernathy, 2021-02-26:
I have my RPI4 booted from a USB SSD.  After booting I plug in a MicroSD card that I was to run Pisafe on.  However, Pisafe can't recognize that SD.  Not sure if the SD has to be using a USB SD card adapter or what?

RichardMidnight, 2021-02-26:
In settings, you can show the root disk.  Disabled by default.

jfabernathy, 2021-02-26:
I see, but when that is enabled, my only choices are /dev/sda which is USB SSD. I never see the internal microSD slot as a choice when I'm booted from USB SSD

jfabernathy, 2021-02-26:
I did get this to work but it's a pain.  I booted from the SD Card I wanted an image of, then plugged in my USB SSD.  I changed the settings of pisafe to allow root backup and I changed the default backup directory to /media/pi/rootfs/home/pi/Download/
It would be great if you could modify your solution so that it works better when you boot the RPI 4 from USB3 SSD and want an image of the internal microSD

RichardMidnight, 2021-02-26:
Oh, I see.  Yes, you figured out how it is designed to work.   I never used it that way, so it did not occur to me.  I was trying to protect users from overwriting their boot disk.   I will consider solutions... Perhaps I need an "enable internal mmc" option. Thanks for the feedback.  
I hope PiSafe is helpful to you.  - Peace

closed 2021-02-26 by jfabernathy

RichardMidnight, 2021-03-12:
New version 0.9.18 should address this. 
Thanks for letting me know