RichardMidnight/pi-safe #9: Some PiSafe GUI recommendations

MardukNT, 2021-03-17:
Hello **RichardMidnight**,
I wanted to suggest some (hopefully) helpful improvements to the GUI for PiSafe.
You are free to ignore any (or all) of the recommendations will be suggesting below, and I won't be offended.

With that said, I will list out the recommended changes, and also attach some screen captures to help illustrate where in the PiSafe GUI I'm referring to.

**Main Menu:**
**1)** I would advise replacing the selectable blank space with a non-selectable horizontal line (see GUI-1.png below).

[image]

**Settings Menu:**
**1)** I would advise replacing the selectable blank space with a non-selectable horizontal line.
**2)** I suggest changing the <'Cancel'> button text to read <'Back'> or <'Back to Main Menu'>, to be less confusing (see GUI-2.png below)

[image]

**Tools Menu:**
**1)** I would advise replacing the selectable blank spaces with non-selectable horizontal lines (or remove them altogether).
**2)** I advise changing "ABOUT PiSafe" to read "About PiSafe" to keep the overall wording consistent on this page.
**3)** I suggest changing the <'Cancel'> button text to read <'Back'> or <'Back to Main Menu'>, to be less confusing (see GUI-3.png below).

[image]

Finally, under the Help Menu, you might want to include an example of how to save a PiSafe image up to a network share.
EG: **pisafe backup /dev/sda /mnt/synology/TwistedOSv20_`date +"%m-%d-%Y_%I%M%p"`.img.gz**

Note: This is the string I'm using in crontab (once a week) to help me capture a backup of my RPi4b, so you would want to design a more generic example for inclusion under the help section.

Thank you again for all your time and efforts in making PiSafe what it is today!

RichardMidnight, 2021-03-19:
MardukMD,
Thanks for the feedback, and I am glad PiSafe is helpful to you.
I have implemented your suggestions with the button labels, caps and example in a beta version.  I actually don't know how to do a non-selectable horizontal line with the tool I am using.  I could remove the spaces completely, but I thought they added clarity by grouping menu items...  I will keep your feedback in mind.

By the way, the beta I am working on has a new feature 'PiSafe lock' which lets you lock a drive so it can't be overwritten in PiSafe.  (Yes, I overwrote my external 2TB drive when I rebooted my pi and it renamed sdb to sda... That was the incentive for 'PiSafe Lock').

MardukNT, 2021-03-19:
RichardMidnight,
Thank you for taking my suggestions to heart. I find your app to be useful and just want to see it shine like the gem it is.

Now, the spaces in and of themselves are not the issue for me. Having them included on the screen to help group things, makes perfect sense.  
But, if you could implement a way for the spaces to be skippable fields (aka non-selectable) as you arrow up/down thru the menu choices, that would be ideal.  If not, so be it.

Either way, keep up the good work!

MardukNT, 2021-03-19:
RichardMidnight,
As a side note, here's another backup solution I've been looking at for my needs: **[framps/raspiBackup](https://github.com/framps/raspiBackup)**
While it looks handy, it constantly pops all kinds of errors whenever I attempt to save a backup up to a network share (a Synology NAS in my case). 

So, after spending long hours trying to troubleshoot a slew of permission/file creation errors, I've decided PiSafe (your app) was the better choice for my backup needs. As it works as advertised.
Just my $0.02 worth.

RichardMidnight, 2021-03-20:
Thanks for the helpful suggestions.  They are implemented in ver 1.0.0.  I am going to leave the spaces in the menu at least for now.

closed 2021-03-20 by RichardMidnight