RichardMidnight/pi-safe #10: Attempting to view PiSafe logs with mousepad generates several deprecation errors

MardukNT, 2021-03-20:
Hello RichardMidnight,
When attempting to view the backup log in **PiSafe**, the terminal window reports the following errors:

(**mousepad**:11129): Gtk-WARNING **: 12:29:06.813: Theme parsing error: <data>:2:29: The style property **GtkButton:default-border** is deprecated and shouldn't be used anymore. It will be removed in a future version

(**mousepad**:11129): Gtk-WARNING **: 12:29:06.813: Theme parsing error: <data>:3:37: The style property **GtkButton:default-outside-border** is deprecated and shouldn't be used anymore. It will be removed in a future version

(**mousepad**:11129): Gtk-WARNING **: 12:29:06.813: Theme parsing error: <data>:4:27: The style property **GtkButton:inner-border** is deprecated and shouldn't be used anymore. It will be removed in a future version

(**mousepad**:11129): Gtk-WARNING **: 12:29:06.813: Theme parsing error: <data>:5:31: The style property **GtkWidget:focus-line-width** is deprecated and shouldn't be used anymore. It will be removed in a future version

(**mousepad**:11129): Gtk-WARNING **: 12:29:06.813: Theme parsing error: <data>:6:28: The style property **GtkWidget:focus-padding** is deprecated and shouldn't be used anymore. It will be removed in a future version

Note: Log file opens in mousepad as expected, so it's working currently. But it may not if/when mousepad gets updated and wanted to call your attention to it.

**Environment:**
PiSafe version: **1.0.0**
OS Mod: **Twister OS version 2.0.0**
OS: **Raspbian GNU/Linux 10 (buster) aarch64**
Host: **Raspberry Pi 4 Model B Rev 1.1**
Kernel: **5.10.17-v8+**
Shell: **bash 5.0.3**
CPU: **(4) @ 2.000GHz**
Memory: **616MiB / 3742MiB**
Disk (/): **11G / 110G (10%)**

RichardMidnight, 2021-03-20:
MardukNT,
Thank you for the feedback.  I have seen these errors.  I figured the Pi foundation would update mousepad sometime or switch to a different editor.  You can specify a different editor in the config file.
Peace

closed 2021-03-20 by RichardMidnight

MardukNT, 2021-03-20:
RichardMidnight,
Per your advice, I've switch PiSafe to use **nano** as my editor of choice, and it works very well. No more errors.
Thank you for your attention to this matter.

--Cheers!