RichardMidnight/pi-safe #11: Help Menu missing a referenced CLI command

MardukNT, 2021-03-20:
Hello RichardMidnight,
When checking the Help Menu, I noticed a CLI command is referenced, but the command itself is absent. (see screenshot).

[image]

RichardMidnight, 2021-03-20:
MardukNT,
Thank you for the feedback.  Typing pisafe without any other arguments brings up the menu.  This is by design, but I was not exactly sure how to note that in the help.  Should I put  < leave blank > ?

MardukNT, 2021-03-20:
RichardMidnight,
Good question.  Might I suggest the following wording:

pisafe    - pisafe <'with no additional arguments'> launches the PiSafe GUI (recommended)

Or something to that effect.
--Cheers

closed 2021-03-20 by RichardMidnight