RichardMidnight/pi-safe #13: [Feature Request] Adding in a choice to review logs using nano instead of using mousepad

MardukNT, 2021-03-20:
Hello RichardMidnight,
Would it be possible to include a choice in PiSafe that could allow the backup logs to be opened with nano instead of mousepad?

I tend to connect to my Raspberry Pi 4 over SSH to kick off backups.  Once backup completes, I find I cannot view the backup log, as PiSafe cannot invoke mousepad while SSH'd in.

Having a prompt/option to view the logs with nano (or some other terminal based editor) during a SSH session, would be very helpful for my needs.

RichardMidnight, 2021-03-20:
MardukNT, Thank you for the feedback.
You can specify any editor you want in the config file.  It is under: Settings, Edit, then the line that says "text_editor="
The current logic is it looks for a number of popular editors and sets to the first one if finds.  If it does not find any, it sets "text_editor" to nano.
Peace.

closed 2021-03-20 by RichardMidnight

MardukNT, 2021-03-20:
Hello RichardMidnight,
Just had a "Doh!" moment after reading this.  For whatever reason, I overlooked manually setting the choice of editor in the PiSafe config file, Now I'm thunking my head on my desk for not trying that first.
As suggested, I changed the value to **nano** and it works as desired.  Thank you for that pearl of wisdom.

--Cheers