RichardMidnight/pi-safe #14: Showing tons of invalid "img" files...

Botspot, 2021-04-07:
Example output of the "LIST" function:
```                                                                                                                                                                                              
    ~~~ IMAGE FILES in '/home/pi/Downloads' ~~~                                                                                                                                                                
    FILE_NAME                             FILE_SIZE  (IMAGE_SIZE)                                                                                                                                              
    ARChon-v2.1.0-arm.zip                 105mb      (239mb)                                                                                                                                                   
    Aster_Chat_Log_(1).zip                1.54mb     (0)                                                                                                                                                       
    Aster_Chat_Log.zip                    860kb      (0)                                                                                                                                                       
    bongocam.zip                          8.00kb     (0)                                                                                                                                                       
    Bongo.Cam.zip                         8.00kb     (0)                                                                                                                                                       
    control.tar.xz                        4.00kb     (0)                                                                                                                                                       
    data.tar.xz                           872kb      (0)                                                                                                                                                       
    FenixPiV0.9.1_4DE_Lite.xz             3.46gb     (0)                                                                                                                                                       
    Github-CLI.zip                        8.00kb     (0)                                                                                                                                                       
    Godot.Engine.3.2.4_(1).zip            8.00kb     (0)                                                                                                                                                       
    Godot.Engine.3.2.4.zip                8.00kb     (0)                                                                                                                                                       
    KSP_demo_win.zip                      464mb      (0)                                                                                                                                                       
    Linux-arm32-AppImage.zip              74.6mb     (0)                                                                                                                                                       
    Lokinet_dscrptn_tun_override_(1).zip  8.00kb     (0)                                                                                                                                                       
    Lokinet_dscrptn_tun_override.zip      8.00kb     (0)                                                                                                                                                       
    Michael_investigation_files.zip       1.83mb     (0)                                                                                                                                                       
    Minecraft.Importer.tar.gz             12.0kb     (0)                                                                                                                                                       
    Notepad.++.zip                        16.0kb     (0)                                                                                                                                                       
    Password.Manager.tar.gz               12.0kb     (0)                                                                                                                                                       
    pi-apps.zip                           15.9mb     (0)                                                                                                                                                       
    Remarkable.zip                        8.00kb     (0)                                                                                                                                                       
    RPi-Play.zip                          8.00kb     (0)                                                                                                                                                       
    Stunt.Rally.zip                       16.0kb     (0)                                                                                                                                                       
    takeout-20210324T180826Z-001.zip      53.3mb     (0)                                                                                                                                                       
    TurboWarp.zip                         8.00kb     (0)                                                                                                                                                       
    Wechat_(1).zip                        12.0kb     (0)                                                                                                                                                       
    Wechat.zip                            12.0kb     (0)                                                                                                                                                       
    zenity.sh.zip                         4.00kb     (0)                                                                                                                                                       
    zoom_i686.tar.xz                      38.5mb     (0) 
```
In addition, it missed all the image files that exist in /home/pi.

RichardMidnight, 2021-04-07:
Thanks for the feedback. 
1)Be sure you are running the latest Ver. 
2)Are you running it from the command line or the menu?

Peace

Botspot, 2021-04-07:
> 1)Be sure you are running the latest Ver.

Yes, I installed it just now. It says it's version 1.0.1.

> 2)Are you running it from the command line or the menu?

Both. It made no difference.  
[image]

RichardMidnight, 2021-04-07:
So, Pisafe is looking for a file ending in ".img" inside the zip or xz file.  If it finds it, it shows the size if it can.  With the menu interface, it looks in the "default directory" set in "settings".  With the Command Line Interface, it looks in the current folder.
 
It looks like you have a bunch of non-pisafe zip and xz files in your default folder (probably downloads).  Either move them out of downloads or move all your pisafe image files to a different folder and set it in settings.

Botspot, 2021-04-07:
> So, Pisafe is looking for a file ending in ".img" inside the zip or xz file. If it finds it, it shows the size if it can.

That's the problem. Your program isn't doing that. Only _**one**_ of those files it found contains an img file. All the others don't contain any files ending in .img. Is this behavior expected? Based on what you said, I wouldn't think so.

> With the menu interface, it looks in the "default directory" set in "settings".
> Either move them out of downloads or move all your pisafe image files to a different folder and set it in settings.

Good to know.

RichardMidnight, 2021-04-08:
This is as expected.  It is to avoid hiding a valid image file from the user.  It shows all img, zip, xz, and gz files in the directory.  Then IF it can get the size of a .img file inside it, it displays the size.

This is because if someone renames the ending of a file it would still be a valid image file, BUT it might not list a .img file inside.

closed 2021-04-08 by Botspot