RichardMidnight/pi-safe #15: Hide dvd drives

Botspot, 2021-04-07:
Output of the "LIST" command:
```
    ~~~ SD-CARD DEVICES ~~~  (hide root device = on)                                                                                                                                                           
    /dev/sdb      Generic  STORAGE_DEVICE       29.7G                                                                                                                                                          
    /dev/sdc      WDC WD10 EALX-759BA1         931.5G                                                                                                                                                          
    /dev/sr0      PLDS     DVD+-RW_DH-16ABS      6.9G                                                                                                                                                          
    /dev/pktcdvd0                                6.9G 
```
/dev/sr0 and /dev/pktcdvd0 are dvd drives. They don't actually contain any dvds at the moment.

I doubt anyone will ever want to use your tool to copy tata from a dvd drive, so would you consider hiding these devices?
In my scripts, I use this:
```
lsblk -I 8,179 -dno PATH,SIZE
```

RichardMidnight, 2021-04-12:
Botspot,

Thank you very much for the feedback and suggestion for filtering block devices.  I think I will implement this in a future release.  I did not understand the MAJ:MIN fields previously, but you inspired me to look into it, looks useful.
peace

RichardMidnight, 2021-04-22:
Should be resolved in v1.0.2

closed 2021-04-22 by RichardMidnight