RichardMidnight/pi-safe #16: Samba?

ST33LDI9ITAL, 2021-04-08:
Hi there, first I'd just like to say thank you for making sd/pi-safe. They are awesome.

Will it work with copying root/active distro to a remote samba share?

RichardMidnight, 2021-04-08:
I think it will work to a samba share... really any available folder.  

The pisafe.tmp file is a crude way to see if you are running the current version.  It downloads pisafe from the GitHub to pisafe.tmp, then runs it to get the most current version. I suppose I should delete automatically ... 

Peace...

MardukNT, 2021-04-09:
@ST33LDI9ITAL 
From personal experience, yes it is possible to backup to a remote **samba/cifs** share using PiSafe.
I currently use PiSafe to backup the 'root/active distro' running on my RPi-4b, up to a Synology NAS shared folder.

Here's an example of what switches I use in **`/etc/fstab`** on the Pi, to help automount the remote samba/cifs share: **`//IP.OF.MY.NAS/RPi4-Backups /mnt/synology cifs vers=3.1.1,_netdev,credentials=/home/pi/.nascreds,uid=pi,gid=pi,x-systemd.automount 0 0`**
**Note**: The `credentials=/home/pi/.nascreds` switch is pointing to a text file that contains the two strings of: '**username=**' and '**password=**' which are needed for authenticating to the samba/cifs share on the Synology.
Note: Change string values to suit your needs.

For PiSafe itself, I'm using this command string to help me capture a backup of the core/active OS on demand and save it up to the network. **pisafe backup /dev/sda /mnt/synology/RPiOSLite-BU_date +"%m-%d-%Y_%I%M%P".img.gz**
Note: Change string values as needed.

Hope that helps.

ST33LDI9ITAL, 2021-04-09:
Got it goin, love it!

RichardMidnight, 2021-04-22:
Should be resolved in v 1.0.2

closed 2021-04-22 by RichardMidnight