RichardMidnight/pi-safe #17: Request

yoshikidneo, 2021-06-28:
Love the tool as is but wondered if there was a way to be able to utilize it for the current SD/SSD img being ran. So instead of having a spare card lying around that you boot up to backup your main SD/SSD, you can run the process for the current drive. If this is already possible, I just don't know how to do it then.

RichardMidnight, 2021-06-28:
Turning off "hide root backup" will allow you to backup the current running root device.  You must back it up to a different device.  Reduce activity on the system as much as possible. I think this is what you are wanting to do.

Peace.

yoshikidneo, 2021-06-29:
Oh wow, I totally looked over that in the options menu. Thanks for pointing it out!

closed 2021-06-29 by yoshikidneo