RichardMidnight/pi-safe #20: Backup to samba share ?

TotalCoolness, 2021-08-31:
Is it possible to backup a running system to a windows samba share (then use that backup and write it to a bootable USB) ?

RichardMidnight, 2021-09-01:
Yes, likely, but I have not tried it, and it may not be reliable.    Shut down as much activity on the system as you can before the backup.
1) If you have a file path to the samba share and enough rights, then you can set the default directory to that filepath.
2) By turning off "hide root device" you can backup the running system.  However, any files being changed by the system will be in an unknown state.  I have successfully backed up the running system, but I have only done it after a fresh reboot so there is not much running.
3) Any backup should be writable to a USB.  The USB will be made bootable when the backup is written to it.  However... your Pi must be set to allow USB boot.  I think there is a firmware upgrade to pi4s that allow this.

TotalCoolness, 2021-09-01:
Yes thanks, the only issue I can see is how do you exclude the samba path from being in the backup ? So for example i.e. /backupspath is the samba share and if we backup the running system surely it will try to include that path as well and we cant unmount it cos we want the backup to be saved on the samba share.

Currently my system boots up fine off NVME USB already so I know that part of the problem has been overcome.

RichardMidnight, 2021-09-01:
I think that is not a problem because pi-safe is copying the bits from the block device (/dev/sda or so) and NOT actual files.  This is a big part of why any changes to the drive during the backup can cause corruption.  It is best to backup a drive that is not in use.

TotalCoolness, 2021-09-02:
Well I tried it, it read 256GB in 1 hour, shrunk to 16GB and save on samba share.

Booted with an SD card and tried the restore to another spare NVME drive, it didnt boot.

Also tried using windows version of raspberry pi imager and same thing no boot. No messages, it just cant boot the drive.

TotalCoolness, 2021-09-04:
I think it could be because it was originally converted from a noobs. Ill play around with it to see if I can work out why.

RichardMidnight, 2021-09-04:
TotalCoolness, I just tested it and it worked.  
I am using a Raspberry Pi 400 and an OMV SMB server.
1) I flashed an 8GB SD card with Raspberry pi OS, got it up and running and created a connection to my SMB server
2) Installed PiSafe, changed the default dir to the SMB server, turned off "hide root device"
3) Using PiSafe, backed up the root drive to the SMB server
4) Put a 16GB USB stick in the Pi
5) Restored the image from the SMB server to the USB stick
6) Rebooted with only the USB stick installed.  The Pi booted, the partition resized to 16GB.

TotalCoolness, 2021-09-04:
Good to know it boots up. Could it be cos I booted up via USB and saved it on the Samba share versus your way of booting via SD card ? Thats the only difference I see.

RichardMidnight, 2021-09-04:
I just booted from the USB and made a new backup of the live system to SMB.
Then still booted from USB I restored to a second USB stick.
That second USB booted fine.

TotalCoolness, 2021-09-04:
Thanks for checking that. My theory is that I  used to have noobs on it and I converted to the standard 2 partitions using a script on the rapsiberry pi forums and some how it has strange partition table now and this affects backups somehow. 

This is the script (usb-boot) on the 1st post. https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=196778 another backup program also has trouble booting sometimes.

RichardMidnight, 2021-09-04:
Yes, I have had enough problems with noobs in the last few years that I don't use it anymore.

closed 2021-09-04 by RichardMidnight