RichardMidnight/pi-safe #22: auto-expand of file system on restore stuck in a reboot loop because of overlay file system

steve-71, 2021-10-17:
discovered an issue where I backed up a system which was configured with overlay file system and read-only boot partition, it shrank it and when I restored it to a new SD Card it tried to auto-expand the filesystem to the media size, rebooted and then tried to expand again.. I couldn't figure out how to get out of this loop, so backed it up again but deselected auto-expand, then the restored image booted correctly.
I could then expand the filesystem after manually disabling the overlay filesystem and read-only partition.
Probably just needs a mention/caution in the docs, or if you can advise how to break out of the reboot loop I'll reproduce the issue.

steve-71, 2021-10-18:
Did a diff between the /boot/config.txt files and there's this line at the end of the one that's using the overlay filesystem:

`initramfs initrd.img-5.10.63-v7l+-overlay`

and did a comparison of the /boot/cmdline.txt and the non-autoexpanding one says: 

`boot=overlay console=tty1 root=PARTUUID=fb514052-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait`

while the normal one omits the `boot=overlay` part, haven't found where it mounts the boot partition read-only yet.
Steve

RichardMidnight, 2021-10-18:
Thanks,
No clues in the fdisk output.  That is normal.  The fact that the last partition is primary/ext4 means it can likely be shrunk.  You are on the right track trying to figure out where it loads the overlay system as read-only.  What is your hardware and OS including ver?  Is this a stock configuration or specialized?
Richard

steve-71, 2021-10-18:
OS details according to pisafe:
`Raspbian GNU/Linux 10 (buster) hw=armv7l kernel=5.10.63-v7l+`
Details from `dmesg`:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1459 SMP Wed Oct 6 16:41:57 BST 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
```

it's Raspberry PI OS Lite (32 bit) according to the Raspberry PI Imager, then update/upgrade'd, so nothing really special going on afaik. I use `sudo raspi-config` performance settings to turn on the overlay filesystem/read only boot partition so it can be power-cycled without shutdown.

I'm installing the beta of pisafe and try that out; thanks.

steve-71, 2021-10-19:
I've tested the new beta; the warning and default of turning off auto-expand works well.
Many thanks for the quick response and fix, most appreciated.

closed 2021-10-19 by steve-71