RichardMidnight/pi-safe #24: CLI backup issues

DmytroKorniienko, 2022-01-23:
1. If curent folder differ then destination then backup will not work (free space issue)

> root@rpi4:/home/kdn# pisafe list
> --- CURRENT SYSTEM ---
> PiSafe ver 1.2.0
> Raspbian GNU/Linux 11 (bullseye) hw=armv7l kernel=5.10.92-v7l+
> Root device = /dev/mmcblk0
> Size      = 29.7gb
> FreeSpace = 20.4gb
> TrashSize = 0
> RootTrash = 0
> 
> 
> --- STORAGE MEDIA (hide root device = ) ---
> /dev/sda     DISAIN   Dual_Flash_Drive 234.4G
> /dev/mmcblk0                            29.7G
> 
> 
> --- IMAGE FILES in '/home/kdn' (20.4gb free) ---
> FILE_NAME  FILE_SIZE  (IMAGE_SIZE)
> root@rpi4:/home/kdn# cd /media/USBFD256G/
> root@rpi4:/media/USBFD256G# pisafe list
> --- CURRENT SYSTEM ---
> PiSafe ver 1.2.0
> Raspbian GNU/Linux 11 (bullseye) hw=armv7l kernel=5.10.92-v7l+
> Root device = /dev/mmcblk0
> Size      = 29.7gb
> FreeSpace = 185gb
> TrashSize = 0
> RootTrash = 0
> 
> 
> --- STORAGE MEDIA (hide root device = ) ---
> /dev/sda     DISAIN   Dual_Flash_Drive 234.4G
> /dev/mmcblk0                            29.7G
> 
> 
> --- IMAGE FILES in '/media/USBFD256G' (185gb free) ---
> FILE_NAME  FILE_SIZE  (IMAGE_SIZE)
> root@rpi4:/media/USBFD256G#

2. Compression set to level  of 9 - what is compression level, why it empty?

> root@rpi4:/media/USBFD256G# pisafe backup /dev/mmcblk0 /media/USBFD256G/rpi4backup/backup_$(date +%Y-%d-%m_%I%M%p).img.gz
> Starting PiSafe 1.2.0 Backup...
> IN-DEV='/dev/mmcblk0'
> OUT-FILE='/media/USBFD256G/rpi4backup/backup_2022-23-01_0427PM.img.gz'
> Compression set to level  of 9
> PiSafe: WARNING 3: SD-card '  - 29.7G (/dev/mmcblk0)' is root device '/dev/mmcblk0'.
> PiSafe: WARNING 4:  '  - 29.7G (/dev/mmcblk0)' is larger than 16GB.
> 
> Use a smaller SD card if you can.
> It will be faster because the entire SD card must be read before it can be compressed.
> 
> A good 8gb card with RaspiOS in a Raspberry Pi 4 takes about 10 minutes
> Step 1 of 3 - Reading  '/dev/mmcblk0' to '/media/USBFD256G/rpi4backup/backup_2022-23-01_0427PM.img' ...
> Sun 23 Jan 2022 04:27:35 PM EET
> Media size=29.7gb
> Skipping 0 of freespace at end of media.
> Reading=29.7gb
> 1.56GiB 0:01:02 [34.6MiB/s] [========>

RichardMidnight, 2022-01-23:
Thanks for the feedback. I will try to fix soon.  
Both problems are probably due it it looking for values in the config file which might not exist.  I guess the CLI does not automatically create the config file.  Can you try running PiSafe with the menu, then when that works you should be able to use the CLI.  Just enter pisafe in a terminal.  That should make the config file.  Compression level should default to 1 which is fastest, it is probably looking for it in the config file. 

Peace

RichardMidnight, 2022-01-24:
DmytroKorniienko,
I found a bug where the compression level is not set in the CLI.  Ver 1.2.1 (beta) should address that.

DmytroKorniienko, 2022-01-25:
@RichardMidnight 
Thanks for your work. I already done all necessary changes for my cron task (redirect to target & run) and issue made only for some improving your great project. Thanks a lot.

Wish you a best.

closed 2022-02-04 by RichardMidnight