RichardMidnight/pi-safe #25: Script does not allow saving to external mounted drive

scetu, 2022-02-06:
Hi, I would like to save backup to mounted external drive `/srv/dev-disk-by-uuid-1ff0486f-334e-4abc-b2ee-eb347354bfad` but pi-safe checks only free space on `/root`, and does not check if directory is mounted drive with different size.

Is there possibility to do this?

Thanks for your effort!

[image]

RichardMidnight, 2022-02-06:
scetu,
Thank you for the feedback.  
I have an SMB server and do not get the same behavior, but I will look into it.  I may just install an option to skip size-checking.
In the meantime, I see that your sdb1 is 214g but only 5% in use.  If you use gparted and resize the partition to 20g or 30g it will probably work and it will work MUCH FASTER because pisafe will ignore the freespace at the end of the media.
Peace.

RichardMidnight, 2022-02-07:
Try the beta version.  In the beta I changed the STOP to a WARNING if the media is too big for the destination.  It should warn you and then continue.
Peace

scetu, 2022-02-07:
Hi,

thank you it works now, I am aware of long running time, but it will run over night, so not a problem for me.
[image]

scetu, 2022-02-07:
Hi,

once more, I would like to say thank for your effort, this is exactly what I've been looking for, but could not find correct software for it.

Here is just complete log how long it ran, imaging SSD to HDD both attached via USB took 2 hours.
[image]

Btw, would not be better to have "Thread storage space errors as warnings" in settings instead of having it globally set?

Best regards

RichardMidnight, 2022-02-08:
Hello scetu,
I believe I found and fixed the issue with miss-calculating the available space on the destination folder (your original issue).  The fix is in the beta I loaded a few minutes ago, same ver 1.2.3,
I have also been reworking the messages so they are more consistent between the CLI and the menu, so there probably still some minor bugs.
Would you test the new beta and let me know how it goes?  Also, can you test restoring the image and running to confirm that works.
- peace

closed 2022-02-09 by RichardMidnight