RichardMidnight/pi-safe #28: Any way to use it on Windows (or WSL)?

jeffrey734, 2022-04-21:
Hi, first thanks a lot for your work on that awsome tool.

I have a lot of things in mind and suggestions, but I will first start with my main point.
I tried someway to use Pisafe, but having my pc on windows I have to boot a live linux each time wich makes the process quite slow.
I tried to use it on WSL, it runs, but it can't detect my micro sd card.
I also tried on a virtual machine, but there again I can't make it see my micro sd card.

On WSL I can see my card sd card on /mnt with also all my other drives. I thought it could be a simple path to add when the tool search to list the drives. I hope it could be easy, I guess WSL could be the easiest way to use it on windows for now as it's all bash.

Thanks

RichardMidnight, 2022-04-21:
Hello Jeffry,

I have used PiSafe with virtualbox on windows very successful.
I fired up WSL but I didn't get PiSafe running on it yet.  I don't remember the exact errors, but it seamed like it was missing a lot of basic Linux functions.

RichardMidnight, 2022-04-24:
Hello Jeffry,
OK, I got WSL running on a clean Windows 10.  PiSafe installed OK (PIGZ does not install, but that is OK).  It seams that WSL does not list usb disks the same a a native linux installation does.  For example the "lsblk" command does not show usb devices.  I see some articles of people working on fixes, but they seem to be still in development.  If I find a fix, I will post it here.  If you find a way to connect a USB drive to WSL and have it show up with the "lsblk" command, let me know.  "usbipd-win project" looks promising, but only works with windows 11. 

Richard

jeffrey734, 2022-04-25:
Ok good thanks. Maybe your first try was on WSL 1 , I saw that WSL 2 was supposed to make things more like real linux system, but I'm nos a pro on that. I'm actually on Windows 11. I will try to look to your suggestion when I have free time and will let you know.

Thanks

RichardMidnight, 2022-04-25:
OK, I got WSL 2 running on Windows 11.  I needed some update patch, but that was mentioned in the install error, and it worked OK.
Now I see that WSL 2 does not support USB devices.  This is the root of the problem. The best resource I found is: https://docs.microsoft.com/en-us/windows/wsl/connect-usb.  I did not get this to work.  It seems to be a work-in-progress.
At then end of the day, PiSafe is expecting the usb drive to be listed when you type in "lsblk".

jeffrey734, 2022-05-04:
Hi, thanks for your effort you're fast. I see what you mean. I did some research too and probably found the same post you found that redirected me on the same projet on the Microsoft blog. I guess the good news there is that seems to be an official project and the team seems to be active.

I tried the install and sucessfully got my SD on my lsusb list using a usb adapter, but not with my SD card port. But it seems that it can't mount it yet as it dosen't appear using lsblk. 

I guess we'll have to let them work a bit more and come back here later.
I'll keep an eye on it on my free time and go on my next suggeestion.

RichardMidnight, 2023-03-28:
I see some progress on usb support for wsl, but still not working for pisafe.  Following the links below, I got "lsusb" to show the usb stick, but "lsblk" still does not.  
For simplicity, I am running windows 11 (updated), wsl 2, and Ubuntu.

https://learn.microsoft.com/en-us/windows/wsl/install
https://learn.microsoft.com/en-us/windows/wsl/connect-usb

RichardMidnight, 2026-09-26:
Hello Jeffry,

Sorry for the long silence — I have been reading this thread and the situation has
changed quite a bit since your last post:

* WSL USB support is now Microsoft's own documented flow, built on the usbipd-win
  project (a Microsoft project, actively maintained):
  https://learn.microsoft.com/en-us/windows/wsl/connect-usb
  https://github.com/dorssel/usbipd-win
* It no longer has the Windows 11 / manual-kernel limitations it had in 2022 — it
  supports Windows 10 and 11 (x64/ARM64) with WSL 2, and it ships through winget.
* A USB *storage* device that is attached this way comes up inside WSL as a normal
  /dev/sd* block device (e.g. /dev/sdb), so it shows up in "lsblk". And that is all
  PiSafe needs: the device list is built from "lsblk" (see DEVICES= in the script),
  so if the card is visible there, PiSafe should work as-is — no code changes
  needed on my side.

To try it again:

1. In Windows PowerShell (as administrator), install the tool if you don't have it:

       winget install usbipd

2. Plug in the SD card reader, then list the USB devices and note the bus ID:

       usbipd list

3. Share (bind) the reader — this step is persistent and only needs to be done once:

       usbipd bind --busid <bus ID>

4. With a WSL terminal open (to keep the WSL 2 VM running), attach it:

       usbipd attach --wsl --busid <bus ID>

5. Inside WSL, check that it appeared:

       lsblk        (and lsusb)

6. If it is listed, run PiSafe as usual.

A few notes:
* This requires WSL 2, not WSL 1. If the device does not appear, update the WSL
  kernel first: "wsl --update" in PowerShell.
* The USB card reader/adapter is the reliable route; the built-in card slot was
  not always seen from the Windows side.
* Binding is persistent; attaching is not — re-attach after reboots.

If the card now shows up in "lsblk" and PiSafe can back up and restore with it,
we can close this issue as solved. If it still does not appear, could you share
the output of "usbipd list" (Windows) and "lsblk -a" plus a few lines of "dmesg"
(inside WSL) after plugging the reader in? I would like to look at it.

Thanks for keeping this thread alive over the years.