RichardMidnight/pi-safe #29: Support of ZStandard compression

jeffrey734, 2022-05-04:
Hi, now that Raspberry Pi Imager imager support the .zst compression as direct input image to flash a SD card, it would be nice to add this compression on ultra(22) which gave me better result on compression.

At this moment I have to use the "no compression" option and then compress my .img file with " zstd" or the 7-Zip ZStandard fork.

I hope that could be easily added.

Thanks

RichardMidnight, 2022-05-06:
Hello Jeffrey734,
Thank you for your feedback.  I believe .zst support can be added to PiSafe.  
I am glad to hear that using the "no compression" option works for you for now.  That is precisely why I put it in.
I will take a look at adding .zst.
 - peace

RichardMidnight, 2022-05-19:
Hello Jeffrey734,

The latest beta (1.2.5.m) supports .zst.  You can edit the settings file (pisafe settings) and change the compression_level line to "compression_level=22 --ultra" with no quotes to get ultra 22.  

I ran a bunch of benchmarks and found xz (9 -e) better on a pi4 with 8gb ram.  See below:

------------------------------------
    Creating image from new raspi-os image expanded on an 8GB SD card.  
    Origional size=837mb
    
    Compression         SIZE        TIME (min:sec)
    
    img                 3.9gb       4:30   
    
    zip 1               1.37g       8:16
    pigz 1              1.37g       6:59         
    xz 1                1.02g       12:54
    zst 1               1.34g       6:28
    
    zip 5               1.28g       10:27
    pigz 5              1.28g       7:15
    xz 5                912m        28:21
    zst 5               1.17g       8:35
    
    zip 9               1.26g       23:04
    pigz 9              1.26g       10:28
    xz 9                849m        35:03
    zst 9               1.13g       10:15
    
    pigz 9 --best        1.26g       11:07
    pigz 11 --best      1.26g       10:43
    xz 9 -e                 848m        43:17           
    zst 22 --ultra       922m        48:22

-- peace