RichardMidnight/pi-safe #30: Problem shrinking Raspi-Endeavouros SDCard.

curyjorge98, 2022-06-30:
Hi Richard,

Are you all right?

I'm still using "PiSafe 1.2.5 i". Is this the right choice for me (remember, brazilian portuguese)?

Today I'm trying to backup a new OS, Endeavour OS.

It has a different way to get the OS on SDCard. By this reason I'd like to have a backup.

But shrinking is failing and I'm not sure restoring would be ok.

The difference I can see is that partitions are GPT. 

Let me know what is going on.

Thanks and see you.

Take a look on fdisk:

Disco /dev/sdd: 29,72 GiB, 31914983424 bytes, 62333952 setores
Disk model: STORAGE DEVICE  
Unidades: setor de 1 * 512 = 512 bytes
Tamanho de setor (lógico/físico): 512 bytes / 512 bytes
Tamanho E/S (mínimo/ótimo): 512 bytes / 512 bytes
Tipo de rótulo do disco: gpt
Identificador do disco: 61DFB242-2BB9-426E-B6A6-3F292848CE81

Dispositivo Início      Fim  Setores Tamanho Tipo
/dev/sdd1     4096   413695   409600    200M Microsoft dados básico
/dev/sdd2   413696 62330879 61917184   29,5G Linux sistema de arquivos

-----------------------------------------------------------------------------------------------------

Take a look on pisafe output:

PiSafe 1.2.5 i
Checking for PiSafe updates...
PiSafe 1.2.5 i Backup... 
No extension specified.  Adding '.img.zip'
OUT-FILE='/home/vieira/Downloads/ISOs/endeavouros-basic.img.zip'
29.7gb to read from '/dev/sdd'
113gb available space on '/home/vieira/Downloads/ISOs' 

~ PiSafe 1.2.5 i Backup '/dev/sdd' to '/home/vieira/Downloads/ISOs/endeavouros-basic.img.zip'
~ Step 1 of 3 - Copying  '/dev/sdd' to '/home/vieira/Downloads/ISOs/endeavouros-basic.img' ... 
qua 29 jun 2022 21:12:12 -03
Media size=29.7gb
Skipping 1.50mb of freespace at end of media.
Reading 29.7gb...
~ Running: sudo pv /dev/sdd -p -s 31913410560  | sudo dd bs=4194304 of='/home/vieira/Downloads/ISOs/endeavouros-basic.img' count=7609 iflag=fullblock conv=fsync 
7609+0 registros de entrada==============================================> ] 99%
7609+0 registros de saída
31914459136 bytes (32 GB, 30 GiB) copiados, 630,515 s, 50,6 MB/s
[========================================================================>] 100%
Done copying media '/dev/sdd'
'30G /home/vieira/Downloads/ISOs/endeavouros-basic.img'
Step 1 took 10 min 36 sec

~ Step 2 of 3 - Shrinking filesystem ...
~ Running: sudo pishrink.sh -s '/home/vieira/Downloads/ISOs/endeavouros-basic.img' 
pishrink.sh v0.1.2
pishrink.sh: Gathering data ...
Error: end of file while reading /home/vieira/Downloads/ISOs/endeavouros-basic.img
pishrink.sh: ERROR occurred in line 281: parted failed with rc 1
pishrink.sh: Possibly invalid image. Run 'parted /home/vieira/Downloads/ISOs/endeavouros-basic.img unit B print' manually to investigate ...
~ Exit_Status 6 running 'sudo pishrink.sh -s '/home/vieira/Downloads/ISOs/endeavouros-basic.img' '
~ Continuing without shrinking the file system...
Done shrinking filesystem.
'30G /home/vieira/Downloads/ISOs/endeavouros-basic.img'
Step 2 took 0 min 5 sec

~ Step 3 of 3 - Compressing '/home/vieira/Downloads/ISOs/endeavouros-basic.img' to '/home/vieira/Downloads/ISOs/endeavouros-basic.img.zip' ... 
qua 29 jun 2022 21:22:53 -03
Compression set to level 5 of 9
29.7gb to compress.
Each dot=608m
                    /home/vieira/Downloads/ISOs/endeavouros-basic.img [........................|.........................] 100%
~ Running: zip -dbds 608m -m -5 '/home/vieira/Downloads/ISOs/endeavouros-basic.img.zip' '/home/vieira/Downloads/ISOs/endeavouros-basic.img'
[   0/ 29G]   adding: home/vieira/Downloads/ISOs/endeavouros-basic.img .................................................. (deflated 68%)
Done compressing '/home/vieira/Downloads/ISOs/endeavouros-basic.img' to '/home/vieira/Downloads/ISOs/endeavouros-basic.img.zip' 
9,6G /home/vieira/Downloads/ISOs/endeavouros-basic.img.zip
Step 3 took 19 min 54 sec
Press any key to return

curyjorge98, 2022-06-30:
Hi Richard,

One more thing...

If you need to test, these are the best instructions, the one I've followed:

https://raspberrytips.com/endeavouros-installation/

Thanks once more,

RichardMidnight, 2022-06-30:
Step 1 did not get a good image of the SD card.  Either one of the two suggestions below may fix the problem.

1) Try using 1.2.5.m. (current beta).  It may have an improvement you need.

2) In the Settings file, set "skip_freespace=off".  There could be a bug in that new feature.

3) If neither of those work, your media (SD card) may be corrupted.  

Richard

curyjorge98, 2022-07-04:
Hi Richard,

Ok, I followed your instructions, made many tests and the conclusion is that there was a temporary corruption in SDCard.

Very good idea to skip non allocated space.

Thanks for your attention.

Keep safe.

Cury

closed 2022-07-04 by curyjorge98

curyjorge98, 2022-07-07:
Hi Richard,
I think it was a temporary corruption in SDCard. I made a new build in a brand new 16 Gb SDCard and pisafe worked like a charm.

In the old 32 Gb I wrote a RPIOS and  pisafe worked like a charm. Well RPIOS is very small and maybe it didn't get the corruption, that's why I sad it was a temporary corruption in SDCard.

I think "skip_freespace=on" is not the problem.

Keep on doing great tools.

Stay safe,