RichardMidnight/pi-safe #32: USBoot Backup?

RobotsAreCrazy, 2023-01-23:
Just wondering will this also allow backup of a boot usb drive as I don't use a microsd to be backedup an external storage e.g usb drive?

RichardMidnight, 2023-01-23:
Yes it should.  It actually creates an image file of any disk-type media.

RobotsAreCrazy, 2023-01-24:
[image]

Any idea, formatted storage to ext4, mounted storage to link in the directory folder?

RichardMidnight, 2023-01-24:
Thanks for the screen shots.  That helped a lot.
I can't see what media you picked to backup.  Is it sda?  Looks like sda has 110g on it.
It looks like you are backing up to sdb which has 444g of space. 
Backing up sda to sdb should work, but it is best to not backup your root drive.  See below.
If you are trying to backup sdb to sdb it won't work.

1) It is better to boot from a different device than you are backing up.  For instance, you could put a minimal linux on what is now sdb, boot to it, install pisafe on it, stick what is now sda in a usb port and backup new-sdb to a folder on new-sda.

2) As an option, you can speed up the backup by shrinking what is not sda2.  you are only using 3% of it.  you can probably use gparted to shrink it.  THIS IS NOT REQUIRED, JUST AN OPTION.

3) Also, leaving the filename ending in zip will compress the image, usually to about half.

let me know if this info helped.

-- peace

RobotsAreCrazy, 2023-01-24:
It's sdb 500gb external SSD

RichardMidnight, 2023-01-24:
OK, to backup a 500gb external SSD (sdb) you need another storage device slightly bigger than 500gb, like a 1TB sdc.  Then backup sdb to sdc.  Again, if you shrink the ext4 partition on sdb first, it will not need as much space or time.

RobotsAreCrazy, 2023-01-24:
Ok, this is my setup I have Debian on 128gb usb drive, I want to image it to the 500gb external SSD

RichardMidnight, 2023-01-24:
that will work.  it is not ideal to backup a root drive, but here is how:
Go into settings, options and turn OFF "hide_root_device".
Then back at the main menu, backup, **sda**, give it a filename (recommend leaving the .zip at the end)

RobotsAreCrazy, 2023-01-24:
> that will work.  it is not ideal to backup a root drive, but here is how:
> Go into settings, options and turn OFF "hide_root_device".
> Then back at the main menu, backup, **sda**, give it a filename (recommend leaving the .zip at the end)

Why is it not ideal?

RichardMidnight, 2023-01-24:
Because the operating system may (or probably will) change some files while the backup is being made. Then the image file is not identical to the original.  
Go ahead with it for now to see it at least starts.  Then consider the process in #1 above.
It is recommended that you backup a non-booted device (sdb) to your booted device (sda).

RobotsAreCrazy, 2023-01-24:
> Because the operating system may (or probably will) change some files while the backup is being made. Then the image file is not identical to the original.
> Go ahead with it for now to see it at [...]

I just wanted to test if it was possible

RichardMidnight, 2023-01-24:
Great. It should work.  let me know how it goes.

RobotsAreCrazy, 2023-01-24:
> Great. It should work.  let me know how it goes.

It's not ideal but I don't have spare storage drives, unless I just use a microsd card

RobotsAreCrazy, 2023-01-24:
PiSafe ver 1.2.6   === Raspberry Pi Imaging App ===                                       PiSafe ver 1.2.6   === Raspberry Pi Imaging App ===                                       PiSafe ver 1.2.6   === Raspberry Pi Imaging App ===                                       PiSafe ver 1.2.6   === Raspberry Pi Imaging App ===                                       PiSafe ver 1.2.6   === Raspberry Pi Imaging App ===                                         GNU nano 7.2              /home/bhadmin/.config/pisafe/pisafe.log                       2023-01-24 14:13 Installing pishrink...                                                   2023-01-24 14:13 Installing pv...                                                         2023-01-24 14:13 Installing PiSafe v 1.2.6...                                             2023-01-24 14:13 ~ PiSafe 1.2.6 installed. Select it from the 'Accessories' menu or type >2023-01-24 14:14 PiSafe 1.2.6 Backup...                                                   2023-01-24 14:15 ~ Error at line 1405. Outfile '' can't be blank                          2023-01-24 14:19 PiSafe 1.2.6 Backup...                                                   2023-01-24 14:20 ~ Error at line 1366. Not enough free space on '/usr/pi-safe' 14.4gb to >2023-01-24 14:21 ~ Error at line 1405. Outfile '/usr/pi-safe/2023-01-24-pisafe.img.img' c>2023-01-24 14:22 Setting PiSafe 1.2.6 to Factory Defaults                                 2023-01-24 14:23 PiSafe 1.2.6 Backup...                                                   2023-01-24 14:24 ~ Error at line 1405. Outfile '' can't be blank                          2023-01-24 14:24 Checking for PiSafe 1.2.6 updates...                                     2023-01-24 14:24 ~ PiSafe 1.2.6 is up to date. Server ver is '1.2.6'                      2023-01-24 14:27 ~ Error at line 2689. ERROR Directory '/usr/pi-safe/images' does not exi>2023-01-24 14:27 ~ Error at line 2689. ERROR Directory '/usr/pi-safe/images' does not exi>2023-01-24 14:31 PiSafe 1.2.6 Backup...                                                   2023-01-24 14:32 ~ Error at line 1366. Not enough free space on '/usr/pi-safe' 14.4gb to >2023-01-24 15:01 ~ Error at line 1366. Not enough free space on '/usr/pi-safe' 14.4gb to >2023-01-24 15:01 ~ Error at line 1405. Outfile '/usr/pi-safe/2023-01-24-pisafe.img' can't>2023-01-24 15:02 ~ Error at line 2767. No media found.                                    2023-01-24 16:38 PiSafe 1.2.6 Format 'WDC PC S WDC PC SA530 SDATB8Y512G1001 - 476.9G (/de>2023-01-24 16:38 ~ Running: sudo dd if=/dev/zero of=/dev/sdb bs=512 count=4               2023-01-24 16:38 ~ Running: sudo parted -s /dev/sdb mklabel gpt                           2023-01-24 16:38 ~ Running: sudo parted -s -a opt /dev/sdb mkpart ext4 ext4 2M 100%       2023-01-24 16:38 ~ Running: sudo mkfs -t ext4 -L ext4 -E root_owner=1000: -F /dev/sdb1    2023-01-24 16:38 ~ Format complete. 'WDC PC S WDC PC SA530 SDATB8Y512G1001 - 476.9G (/dev>2023-01-24 16:39 ~ Running: udisksctl mount -b /dev/sdb1                                  2023-01-24 16:39 Checking for PiSafe 1.2.6 updates...                                     2023-01-24 16:39 Checking for PiSafe 1.2.6 updates...                                     2023-01-24 16:40 Checking for PiSafe 1.2.6 updates...                                     2023-01-24 17:02 PiSafe 1.2.6 Backup...                                                   2023-01-24 17:04 ~ Error at line 1366. Not enough free space on '/usr/pisafe-backups' 476>2023-01-24 17:58 ~ Error at line 1366. Not enough free space on '/usr/pisafe-backups' 476>2023-01-24 18:04 ~ Error at line 1405. Outfile '/usr/pisafe-backups/2023-01-24-pisafe.img>2023-01-24 18:07 ~ Error at line 2689. ERROR Directory '/dev/sdb' does not exist          2023-01-24 18:42 ~ Error at line 2689. ERROR Directory '/dev/sdb/images' does not exist   2023-01-24 20:30 PiSafe 1.2.6 Backup...                                                   2023-01-24 20:31 ~ Error at line 1405. Outfile '' can't be blank                          2023-01-24 20:31 PiSafe 1.2.6 Backup...                                                   2023-01-24 20:32 ~ Error at line 1390. Media '' can't be blank.                                                               [ Read 50 lines ]                                     ^G Help        ^O Write Out   ^W Where Is    ^K Cut         ^T Execute     ^C Location    ^X Exit        ^R Read File   ^\ Replace     ^U Paste       ^J Justify     ^/ Go To Line

RichardMidnight, 2023-01-24:
As I mentioned above, if you use gparted to shrink the ext4 partition on sda, it can go MUCH faster because it will skip all the "free space" you free up.  Maybe resize it from 119GB to 20 or 30GB.  You can always resize it larger later.

RobotsAreCrazy, 2023-01-24:
At the shrink part it had an error, but closed out way too fast for me to screencap it!

RichardMidnight, 2023-01-24:
open the log file.  Tools / Log

RobotsAreCrazy, 2023-01-24:
> open the log file.  Tools / Log

It didn't capture the error for some strange reason

RichardMidnight, 2023-01-24:
If you don't care what is on your external 500gb drive, install reapberrypiOS (or other linux) on it, boot to it, install pisafe on it, plug your 119gb in as an sdb and backup sdb.  Consider taking the defaults of Downloads folder and .zip extension.  Then experiment later.

RobotsAreCrazy, 2023-01-24:
> If you don't care what is on your external 500gb drive, install reapberrypiOS (or other linux) on it, boot to it, install pisafe on it, plug your 119gb in as an sdb and backup sdb.  Consider taking the defaults of Downloads folder and .zip extension.  Then experiment later.

[image]

Managed to get that

RichardMidnight, 2023-01-24:
hmm.  Use raspberry pi imager from their website to flash the OS to the 500gb drive.

RichardMidnight, 2023-01-25:
Any luck with creating an image of your 119gb usb stick?

RichardMidnight, 2023-01-25:
It looks like you are still booting to the 119gb.  You should boot to the 500gb.  then there will be room to backup the 119 to the 500.

RobotsAreCrazy, 2023-01-25:
> It looks like you are still booting to the 119gb.  You should boot to the 500gb.  then there will be room to backup the 119 to the 500.
> 

No I'm booted on the 500gb one now with fresh  install of pi-safe

RichardMidnight, 2023-01-25:
I just have a moment for now.  Will check in later.
Can you remove the 119, boot to the 500, set pisafe to factory defaults, plug the 119 in, select backup...

RobotsAreCrazy, 2023-01-25:
> I just have a moment for now.  Will check in later.
> Can you remove the 119, boot to the 500, set pisafe to factory defaults, plug the 119 in, select backup...
> 

That's what I did...

RichardMidnight, 2023-01-25:
Or use gparted to expand it.

RobotsAreCrazy, 2023-01-25:
> Or use gparted to expand it.

Wish it was that simple, I don't have a desktop environment installed and also I use zramswap...just going to easier with a fresh pi os install on 500gb just used for image backups

RobotsAreCrazy, 2023-01-25:
> Or use gparted to expand it.

Update: 100% worked

[image]

Just ended up flashing new pi os, install pi-safe and will just use, 500GB as an image backup drive

RichardMidnight, 2023-01-25:
Terrific!!  Glad it worked.
You can use the pisafe or raspi-imager to restore the image to a media.
If all went well, you can restore it to any media (sd, usb stick, usb ssd...)
You can choose .xz and even kick the compression level to make a smaller archive... but it will take more time.
You can name the image file to reflect what is in it.  Best to NOT rename the file after it has been made.
- peace

closed 2023-01-30 by RichardMidnight