RichardMidnight/pi-safe #33: Feature Request: integrate gpg encryption of images

MrOnak, 2023-02-03:
This tool is the answer to a long search of mine, so thanks a ton! <3

I have one possible feature request, although I personally classify it as low-priority since it can easily be added manually: I'd like to see added GnuPG encryption of the compressed archives (*.img.zip.gpg and alternatives). 

I do this to have encrypted files be transferred to offsite backup sites that I don't fully trust.

One way to do this would be to ask for a GPG key id at some point in the backup / restore process or via commandline parameter, then run `gpg --encrypt --recipient [KEYID] ~/Downloads/backupfile.img.zip` and remove the .zip file later. 

Alternatively pipe the *zip command into gpg directly and never store the .*zip file on disk in the first place. This is probably faster and preferred by the paranoid since it won't leave unencrypted artefacts on the local harddisk. 

Example for gzip:
```
gzip --stdout - | gpg --encrypt --recipient $GPGKEYID --output $TARGETFILE
```

Thanks for your consideration

RichardMidnight, 2023-02-04:
Hello MrOnak, Thanks for the feedback.  I am glad PiSafe is useful to you.  If you know of any ways to spread the word about PiSafe, I would love to know.
I will look into an option to encrypt the image file.  I cannot focus on it for a while, but it makes sense.
Thanks again
peace

RichardMidnight, 2023-02-15:
One immediate way to do this is to have Pisafe create an img file by having your backup filename end in ".img".  Then pisafe will shrink the filesystem but will NOT compress the file.  So you can run your own compression with encryption. 
gzip --stdout - | gpg --encrypt --recipient $GPGKEYID --output $TARGETFILE
peace

MrOnak, 2023-02-18:
I'm guessing since pi-safe does shrink the image after writing the initial image to disk there won't be a way to avoid having the .img on the disk, correct? If that's the case, then the piping approach I outlined above won't work. Which is probably fine for most people including myself ;)

As I said, adding the `gpg` step manually after the pi-safe run is not a big issue, still adding it might be a convenience feature

closed 2023-03-28 by RichardMidnight