RichardMidnight/pi-safe #34: pigz not working

BoraFreak, 2023-02-11:
~ Running: pv '/home/pi/2023-02-11-pisafe.img' -p | pigz --best= -1 > '/home/pi/2023-02-11-pisafe.img.gz'
pigz: abort: invalid option: --best=
[>                                                                         ]  0%
~ Exit_Status 22 running 'pv '/home/pi/2023-02-11-pisafe.img' -p | pigz --best= -1 > '/home/pi/2023-02-11-pisafe.img.gz''

with --best= -5 or --best= -9 is it the same issue

RichardMidnight, 2023-02-11:
Hello BoraFreak,
Thank you for the feedback.  I was able to recreate the error, so that is good.  I will be working on a fix.
- peace

RichardMidnight, 2023-02-11:
I think the beta version fixes the issue.  Follow the Install instructions for the beta version.

- peace

BoraFreak, 2023-02-12:
thanks for your quick answer

BoraFreak, 2023-02-12:
Beta Version works fine

~ Step 3 of 3 - Compressing '/home/pi/PiOS Lite 32Bit.img' to '/home/pi/PiOS Lite 32Bit.img.gz' ...
So 12. Feb 15:36:46 CET 2023
Compression set to level 9 of 9
2.43gb to compress.
~ Running: pv '/home/pi/PiOS Lite 32Bit.img' -p | pigz -9 > '/home/pi/PiOS Lite 32Bit.img.gz'
[========================================================================>] 100%
Done compressing '/home/pi/PiOS Lite 32Bit.img' to '/home/pi/PiOS Lite 32Bit.img.gz'
959M /home/pi/PiOS Lite 32Bit.img.gz
Step 3 took 8 min 50 sec

 ~ Backup complete.
   │ 'Kingston UHSII_uSD_Reader - 7,4G (/dev/sda)' backed up to
   │ '/home/pi/PiOS Lite 32Bit.img.gz'
   │ 7.40gb reduced to .93gb in 14 min 33 sec.

thank you very much

closed 2023-02-26 by RichardMidnight