RichardMidnight/pi-safe #35: Syntax error on Debian 11.

curyjorge98, 2023-02-13:
Hi Richard,

I've been using Pisafe a lot in Ubuntu, RPI OS and I am sure it is a great tool.

Today I am trying to use PiSafe on my new SBC, Orange PI 5, formatted with Debian 11 - KDE 
and incredibly I am receiving these 3 error messages. 

My problem is solved cause I Still have RPI 4 and PiSafe is working good on it.

Stay safe and healthy,

cury@opi5ssd2:~$ sh ./pisafe.sh 
./pisafe.sh: 455: Syntax error: "(" unexpected (expecting "}")
cury@opi5ssd2:~$ ls -l
drwx------   - cury 12 jan 20:51 Desktop
drwxr-xr-x   - cury 11 fev  0:10 Downloads
drwxr-xr-x   - cury 20 jan  2:52 Imagens
drwx------   - cury 12 jan 22:18 NoMachine
drwxr-xr-x   - cury  9 fev  4:10 Pictures
.rwxr-xr-x 71k cury 13 fev  3:44 pisafe.sh
.rwxr-xr-x 71k cury 13 fev  3:49 pisafe_1.1.1
.rw-r--r-- 85k cury 13 fev  3:49 pisafe_1.2.5
drwxrwx--x   - cury 12 fev 21:25 Videos
cury@opi5ssd2:~$ sh ./pisafe_1.1.1
./pisafe_1.1.1: 455: Syntax error: "(" unexpected (expecting "}")
cury@opi5ssd2:~$ sh ./pisafe_1.2.5
./pisafe_1.2.5: 690: Syntax error: "(" unexpected (expecting "}")
cury@opi5ssd2:~$

RichardMidnight, 2023-02-15:
Hello Curygorge98,
I am glad PiSafe has been working for you, and thanks for the feedback.
I think the issue is that PiSafe is a BASH script not a SHELL script.  try 'bash' instead of 'sh'.
Also, you should usually run the latest published version named pisafe.  
The install instructions shows two lines that download pisafe, then run it with the install command.
Let me know if that helps.
- peace

RichardMidnight, 2023-02-15:
Hello Curygorge98,
I just confirmed with Debian 11, that if you run pisafe with sh instead of bash you will get the error you got.
Peace

curyjorge98, 2023-02-15:
Hi Richard,

Sorry, my mistake.

I am trying to backup a 64gb SDCard formatted GPT.

Will it work???

PiSafe 1.2.5
PiSafe 1.2.5 Backup... 
No extension specified.  Adding '.img.zip'
OUT-FILE='/media/cury/BPK1/Pisafe/2023-02-15-Rebrn-opi5-64g.img.zip'
58.2gb to read from '/dev/sdb'

Thanks,

RichardMidnight, 2023-02-15:
64GB will work as long as you have space on your destination media (/media/cury/BPK1...).  1.5X should be plenty.  It will be slow.  If there is a lot of empty space on your 64GB card, you can use gparted to shrink the last partition.  Then pisafe will skip that part which will speed things up a lot.  Don't shrink all the way, leave some breathing room.

closed 2023-02-16 by RichardMidnight