RichardMidnight/pi-safe #37: On backup, always installing pishink

robex2005, 2023-03-28:
Hello, and thank you for the great script that you created!
I'm using it from cron with a script, with a little problem...

The script that is called from cron (pisafe_backup.sh):

_#!/bin/bash
rm /home/root/.config/pisafe/pisafe.log
/usr/local/bin/pisafe backup /dev/mmcblk0 /mnt/extra/system_backup/backup/shrinked/'rPI_'$(date +%d\%m\%Y)'.img' -y_

The problem is , that on every single run, in the log file, i see that your script, always downloads and install pishrink.sh (even if the file is already downloaded)

2023-03-28 10:00 Installing pishrink...
2023-03-28 10:00 PiSafe 1.2.9 Backup...
2023-03-28 10:00 ~ PiSafe 1.2.9 Backup '/dev/mmcblk0' to '/mnt/extra/system_backup/backup/shrinked/rPI_28032023.img'
2023-03-28 10:00 ~ Step 1 of 3 - Copying '/dev/mmcblk0' to '/mnt/extra/system_backup/backup/shrinked/rPI_28032023.img' ...
2023-03-28 10:00 ~ Running: sudo pv /dev/mmcblk0 -p -s 15552479232 | sudo dd bs=4194304 of='/mnt/extra/system_backup/backup/shrinked/rPI_28032023.img' count=3709 iflag=fullblock conv=fsync
2023-03-28 10:13 ~ Step 2 of 3 - Shrinking filesystem ...
2023-03-28 10:13 ~ Running: sudo pishrink.sh '/mnt/extra/system_backup/backup/shrinked/rPI_28032023.img'
2023-03-28 10:26 ~ Step 3 of 3 - Compressing '/mnt/extra/system_backup/backup/shrinked/rPI_28032023.img' to '/mnt/extra/system_backup/backup/shrinked/rPI_28032023.img' ...
2023-03-28 10:26 ~ Backup complete.  -' - 14.5G (/dev/mmcblk0)' backed up to  -'/mnt/extra/system_backup/backup/shrinked/rPI_28032023.img'  -14.4gb reduced to 6.21gb in 26 min 4 sec.

content of /usr/local/bin:

_root@rPi:/usr/local/bin# ls
google-oauthlib-tool  gphotos-sync  pisafe  pishrink.sh  pyrsa-decrypt  pyrsa-encrypt  pyrsa-keygen  pyrsa-priv2pub  pyrsa-sign  pyrsa-verify
root@rPi:/usr/local/bin#_ 

What is the solution to avoid downloading pishrink every time?

Thanks!

RichardMidnight, 2023-03-28:
I have not tried to reproduce your results yet, buy you can try this:

1) Be sure you are running the latest version.  1.2.9 
2) Open the Settings file: with the menu, or typing "pisafe settings"
3) Change "check for updates on startup" to off
4) you may also want to change "check dependencies on startup" to off.

once you do this, you should run "pisafe update" periodically to keep on the latest version.

let me know if that resolved the issue.

- peace

RichardMidnight, 2023-03-28:
This may be the more precise resolution:

By default, Pisafe runs a "which" on each dependency including pishrink.sh on startup.  If "which" does not find an instance of it, Pisafe installs it.  Can you run "which pishrink.sh" and see if it returns "/usr/local/bin/pishrink.sh".  If not, type "echo $PATH" and see if "/usr/local/bin" is in your path so "which" can search it.
If needed: add "/usr/local/bin" to your path.

RichardMidnight, 2023-03-28:
Very strange.  I just confirmed on my system that it does NOT reinstall pishrink every time.  Pi4 running RaspberryPi-OS.
What is your hardware and OS version?
This is mysterious, but not a big concern.  Reinstalling pishrink is an extremely lightweight operation.

robex2005, 2023-03-29:
I'm running the script on a RPi 3 Model B, Debian GNU/Linux 11 (bullseye), Kernel: Linux 6.1.19-v7+ with DietPi distro

I've modified the setting file, turning off only de dependency part
On running the script, it does not reinstall pishrink...
So , for now, this is a workaround...

RichardMidnight, 2023-03-29:
Hmmm.. I was not able to reproduce your results.  
Glad turning off dependencies worked.
Try resetting pisafe settings file back to factory defaults by typing in 'pisafe defaults'

I downloaded a fresh dietpi (apparently 64bit), added the lxde desktop (just so it would mount the usb media).   I see you are running 32 bit on a pi 3.  I doubt the hardware makes a difference.  Maybe the 32-bit?

My Hardware is: a Pi4
My OS is: root@DietPi:~# uname -a
Linux DietPi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux

Log is: 
2023-03-29 17:36 PiSafe 1.2.9 Backup...
2023-03-29 17:36 ~ Error at line 1399. File './test.img' already exists
2023-03-29 17:36 PiSafe 1.2.9 Backup...
2023-03-29 17:36 ~ PiSafe 1.2.9 Backup '/dev/sda' to 'test2.img.zip'
2023-03-29 17:36 ~ Step 1 of 3 - Copying '/dev/sda' to './test2.img' ...
2023-03-29 17:36 ~ Running: sudo pv /dev/sda -p -s 4387241984 | sudo dd bs=4194304 of='./test2.img' count=1047 iflag=fullblock conv=fsync

RichardMidnight, 2023-03-31:
I was able to get fresh download of dietpi 32-bit running on a pi3b. But was still not able to reproduce your results.
uname -a says:
Linux DietPi 6.1.19-v7+ #1637 SMP Tue Mar 14 11:04:52 GMT 2023 armv71 GNU/Linux

The code is running "which pishrink.sh".  If there is an error code returned it tries to install pishrink.sh.

My suggestion is to:
update linux and reboot
reinstall pisafe, reset settings to default

- peace

robex2005, 2023-04-03:
Linux is up to date, reinstalled pisafe, settings to default, but still not finding pishrink...
Knowing that all the dependencies are installed correctly, in my case, disabling the dependency check works...
Glad to hear that is not a coding problem (probably i have a problem with my live distro - over the years i did quite alot of installing and testing scripts, including in python - probably something is messed up, altho everything is working fine at the moment...)

Thanks for your support and keep up the good work!

RichardMidnight, 2023-04-08:
OK, one last idea before I close this issue.
What do you get when you type in:

which pishrink.sh
echo $?

I get a 0 returned meaning there was no error with "which pishrink.sh".  I expect you are not getting 0 returned.  See below:

pi@RaspberryPi-Master:~ $ which pishrink.sh
/usr/local/bin/pishrink.sh
pi@RaspberryPi-Master:~ $ echo $?
0
pi@RaspberryPi-Master:~ $

robex2005, 2023-04-11:
Hello, 

Sorry for the late reply...

My output:

root@rPi:~# which pishrink.sh
/usr/local/bin/pishrink.sh
root@rPi:~#  echo $?
0
root@rPi:~#