RichardMidnight/pi-safe #38: Time after time

kPlusPlus, 2023-04-21:
1. idea Can you add some estimated time
for create image
and for compress

Or pointing me where in code I can do it.
Thx

RichardMidnight, 2023-04-22:
Hello,  and thanks for the feedback.

1) I put the three progress bars on the screen.
2) I put on how long it takes to create an image of RaspberryPiOS on a Pi4 and an 8gb card.
3) Then it logs the actual time for each job.

Are you looking for it to estimate the time for each job before it runs?  I like the idea... but...
Since PiSafe runs on any linux on any hardware (not just raspberry pi), I am not sure how I would estimate the time.  It would also depend on the size of the sd card and the compression tool and settings.

-peace

RichardMidnight, 2023-05-01:
1) PiSafe tells you "a good 8GB sd card with RaspiOS on a Pi4 takes about 11 minutes".
2) There is a pretty good estimate of the SIZE of the backup before it runs.  
3) The log file tells you the actual size and time of the backup or restore.
It is impossible to estimate the time required unless we do a calibration on each installation.  I am thinking this is out of the scope of this project.

Again, thanks very much for the feedback.  I hope PiSafe is useful to you.

- peace

closed 2023-05-01 by RichardMidnight