RichardMidnight/pi-safe #41: nvme support?

cklann1, 2023-10-23:
Hi - Thanks great piece of software!

With the new rpi5 coming out...and, in support for x86-based systems, is it possible to set up nvme-based systems?  I've successfully run pi-safe on Debian/x86-based systems with SSD's (ie, /dev/sd*1), but when I try to configure on a Debian system with / mounted on /dev/nvme0n1, it doesn't show up in the media screen - STORAGE MEDIA (hide root device = off) .  Not sure if this is an enhancement request or a configuration issue.

Thanks!

RichardMidnight, 2023-10-23:
Thanks for the feedback. I am glad that PiSafe is working for you.

I believe I can enable NVME support.  I may ask you to run a command on your system in the next few days and tell me the results.

RichardMidnight, 2023-10-23:
I think I need the results of "lsblk" when your nvme is up and running. Can you run that for me?

cklann1, 2023-10-23:
Hi - thanks for the quick turnaround!

Here's the lsblk on this Debian 11 / Proxmox machine:
```shell
cklann@dell2:~$ sudo lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop0         7:0    0    10G  0 loop 
loop1         7:1    0     8G  0 loop 
loop2         7:2    0     4G  0 loop 
loop3         7:3    0     8G  0 loop 
sda           8:0    0   1.8T  0 disk 
├─sda1        8:1    0 651.9G  0 part 
└─sda2        8:2    0   1.2T  0 part 
nvme0n1     259:0    0 931.5G  0 disk 
├─nvme0n1p1 259:1    0     1M  0 part 
├─nvme0n1p2 259:2    0   512M  0 part 
└─nvme0n1p3 259:3    0 214.1G  0 part /
``` 
Cheers!

RichardMidnight, 2023-10-23:
I just updated the beta version.  Let me know if that does it.  Make sure you get 1.2.10c or newer

cklann1, 2023-10-23:
Wow - Thanks!  Will check it out in a few hours. 👍

cklann1, 2023-10-24:
Looks like it worked on my setup, tho it looked like pishrink didn't...will try it on another NVME drive soon.
```shell
cklann@dell2:~$ pisafe                                                                                                                                                                                      [16/1170]
PiSafe 1.2.10 c                                                                                                                                                                                                          
Checking for PiSafe updates...                   
PiSafe 1.2.10 c Backup... 
mount: /tmp/tmp.EJW9PYD76I: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
mount: /tmp/tmp.m4ZQROnviX: mount(2) system call failed: File exists.
mount: /tmp/tmp.uYLGV4mhZ0: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
mount: /tmp/tmp.VuM2Q5WoaR: mount(2) system call failed: File exists.                                                                                                                                                    
No extension specified.  Adding '.img.zip'
OUT-FILE='/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img.zip'
214gb to read from '/dev/nvme0n1'
1.67tb available space on '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2'                                                                                                                                              
                                                      
~ PiSafe 1.2.10 c Backup '/dev/nvme0n1' to '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img.zip'
~ Step 1 of 3 - Copying  '/dev/nvme0n1' to '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' ...                                                                                                   
Mon 23 Oct 2023 08:08:15 PM CDT                                                                                                                                                                                          
Media size=.90tb               
Skipping 716gb of freespace at end of media.
Reading 214gb...
~ Running: sudo dd if=/dev/nvme0n1 bs=4194304 count=54947 | pv -s 230462324736  | sudo dd of='/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' conv=fsync 
54947+0 records in138MiB/s] [========================================================================================================================================================================>  ] 99% ETA 0:00:01
54947+0 records out
230464421888 bytes (230 GB, 215 GiB) copied, 3850.72 s, 59.8 MB/s
 214GiB 1:04:10 [57.1MiB/s] [=========================================================================================================================================================================>] 100%            
450125824+0 records in
450125824+0 records out
230464421888 bytes (230 GB, 215 GiB) copied, 3855.9 s, 59.8 MB/s
Done copying media '/dev/nvme0n1'
'40G /mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img'
Step 1 took 64 min 21 sec

~ Step 2 of 3 - Shrinking filesystem ...
~ Running: sudo pishrink.sh  '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' 
pishrink.sh v0.1.3
pishrink.sh: Gathering data ...
Error: end of file while reading /mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img
pishrink.sh: ERROR occurred in line 284: parted failed with rc 1
pishrink.sh: Possibly invalid image. Run 'parted /mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img unit B print' manually to investigate ...
~ Exit_Status 6 running 'sudo pishrink.sh  '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' '
~ Continuing without shrinking the file system...
Done shrinking filesystem.
'40G /mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img'
Step 2 took 0 min 8 sec

~ Step 3 of 3 - Compressing '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' to '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img.zip' ... 
Mon 23 Oct 2023 09:12:44 PM CDT
Compression set to level 1 of 9
39.1gb to compress.
~ Running: pv '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' | zip -1 '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img.zip' - 
zip I/O error: Permission denied
zip error: Could not create output file (/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img.zip)
64.0KiB 0:00:00 [7.69MiB/s] [>                                                                                                                                                                          ]  0%            
~ Exit_Status 15 running 'pv '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' | zip -1 '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img.zip' - '
Getting list of images ready...
``` 

Thanks again!

RichardMidnight, 2023-10-24:
Good, glad that worked.

Thanks for the log.  I think the error in step 3 is a bug I can fix. 

I wonder if the pishrink error is because of the disk being too big.  Not really sure.

cklann1, 2023-10-24:
For grins, I ran using sudo...looks like that makes step 3 work:
```shell
cklann@dell2:~$ sudo pisafe                                                                                                                                                                                 [12/1291]
PiSafe 1.2.10 c                                                                                                                                                                                                          
Checking for PiSafe updates...                   
PiSafe 1.2.10 c Backup... 
mount: /tmp/tmp.uHFDLq549C: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
mount: /tmp/tmp.hqLY35bUpC: mount(2) system call failed: File exists.
mount: /tmp/tmp.D8DSyyZOs3: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
mount: /tmp/tmp.KAWhH8NEgJ: mount(2) system call failed: File exists.                                                                                                                                                    
No extension specified.  Adding '.img.zip'
OUT-FILE='/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img.zip'
214gb to read from '/dev/nvme0n1'
1.67tb available space on '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2'                                                                                                                                              
                                                                                                                                                                                                                         
~ PiSafe 1.2.10 c Backup '/dev/nvme0n1' to '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img.zip'                                                                                                   
~ Step 1 of 3 - Copying  '/dev/nvme0n1' to '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' ... 
Mon 23 Oct 2023 09:30:53 PM CDT
Media size=.90tb
Skipping 716gb of freespace at end of media.
Reading 214gb...
~ Running: sudo dd if=/dev/nvme0n1 bs=4194304 count=54947 | pv -s 230462324736  | sudo dd of='/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' conv=fsync 
54947+0 records in136MiB/s] [========================================================================================================================================================================>  ] 99% ETA 0:00:00
54947+0 records out
230464421888 bytes (230 GB, 215 GiB) copied, 3745.9 s, 61.5 MB/s
 214GiB 1:02:25 [58.7MiB/s] [=========================================================================================================================================================================>] 100%            
450125824+0 records in
450125824+0 records out
230464421888 bytes (230 GB, 215 GiB) copied, 3774.68 s, 61.1 MB/s
Done copying media '/dev/nvme0n1'
'40G /mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img'
Step 1 took 63 min 0 sec

~ Step 2 of 3 - Shrinking filesystem ...
~ Running: sudo pishrink.sh  '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' 
pishrink.sh v0.1.3
pishrink.sh: Gathering data ...
Error: end of file while reading /mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img
pishrink.sh: ERROR occurred in line 284: parted failed with rc 1
pishrink.sh: Possibly invalid image. Run 'parted /mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img unit B print' manually to investigate ...
~ Exit_Status 6 running 'sudo pishrink.sh  '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' '
~ Continuing without shrinking the file system...
Done shrinking filesystem.                                                                                  
'40G /mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img'
Step 2 took 0 min 8 sec
                                                      
~ Step 3 of 3 - Compressing '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' to '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img.zip' ... 
Mon 23 Oct 2023 10:34:01 PM CDT
Compression set to level 1 of 9
39.1gb to compress.                                                                                         
~ Running: pv '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img' | zip -1 '/mnt/pve/nas1-zfsbackups/myshared/pi-safe/dell2/2023-10-23-pisafe.img.zip' - 
  adding: -68.8MiB 0:00:01 [68.7MiB/s] [>                                                                                                                                                                          ]  0% 
35.7GiB 0:14:40 [47.0MiB/s] [==========================>                                                                                                                                                ] 16% ETA 1:13:32
```

RichardMidnight, 2023-10-24:
Good.  That confirms what I suspected, I will add sudo to step 3.  

Is the last partition on /dev/nvme0n1 an ext4 partition?  
Can you run 'sudo parted /dev/nvme0n1 print free" for me?

cklann1, 2023-10-24:
Nope - it's free space...
```shell
cklann@dell2:~$ sudo parted /dev/nvme0n1 print free
Model: CT1000P3SSD8 (nvme)
Disk /dev/nvme0n1: 1000GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size    File system  Name  Flags
        17.4kB  1049kB  1031kB  Free Space
 1      1049kB  2097kB  1049kB                     bios_grub
 2      2097kB  539MB   537MB   fat32              boot, esp
 3      539MB   230GB   230GB   btrfs
        230GB   1000GB  770GB   Free Space
```

RichardMidnight, 2023-10-24:
So pisafe ignores free space at the end.  BUT... pishrink does not work with btrfs.

cklann1, 2023-10-24:
Oh, ok...I
Sometime soon, I'll try restoring that btrfs image to another machine and see what happens

Thanks again!

cklann1, 2023-10-24:
Closing - pisafe nvme now supported! 👍

closed 2023-10-24 by cklann1

RichardMidnight, 2023-10-24:
Glad it works for you!  If you have a lot of empty space in your btrfs partition, I think you can use gparted to shrink it a bit before using pisafe.
- Peace

cklann1, 2023-10-24:
Thanks - yep, gotta do some disk hygiene 🙂

Cheers!

RichardMidnight, 2023-11-03:
Today's release supports nvme.  

Thanks for the feedback.