RichardMidnight/pi-safe #43: Version 1.2.10  Slow

kPlusPlus, 2023-11-09:
New version for my 64Gb Rpi4 image over 4 hours ?!?
This is toooo much. What is wrong ?

RichardMidnight, 2023-11-09:
64GB is pretty big, but that does sound excessive.  
1) I just popped a 64gb usb stick in a pi 4 and pisafe is estimating 1 hour for step 1.
2) I assume this is with the old version too 1.2.9 because I don't think there are any changes that would cause this issue.  If you can test the old version and confirm, that would be great.
3) Is the 4 hours the entire process? or mostly one of the 3 steps?
4) If you can shrink the last partition and leave free space at the end of the drive, Pisafe will ignore the freespace.  gparted is pretty easy to use for this.
5) Make sure you are using USB 3 (blue) drives and plug into the USB 3(blue) ports.  USB2 will be too slow for this size data.

kPlusPlus, 2023-11-09:
[image]

Heh new version use all 223 GB and no shrink.
I must downgrade. uf uf uf

kPlusPlus, 2023-11-09:
How Can I downgrade to 1.2.9 ?!

RichardMidnight, 2023-11-10:
Your dev/sdb is 200gb, did you select the correct device?
Below the install instructions is install for beta , then install for last ver: 1.2.9

RichardMidnight, 2023-11-10:
I just tested a 64gb image.  It took 160 minutes.  

2023-11-09 16:26 ~ PiSafe 1.2.10 Backup '/dev/mmcblk0' to '/media/pi/rsync/2023-11-09-pisafe-test.img.zip'
2023-11-09 16:26 ~ Step 1 of 3 - Copying '/dev/mmcblk0' to '/media/pi/rsync/2023-11-09-pisafe-test.img' ...
2023-11-09 16:26 ~ Running: sudo dd if=/dev/mmcblk0 bs=4194304 count=15227 | pv -s 63864569856 | sudo dd of='/media/pi/rsync/2023-11-09-pisafe-test.img' conv=fsync
2023-11-09 17:23 ~ Step 2 of 3 - Shrinking filesystem ...
2023-11-09 17:23 ~ Running: sudo pishrink.sh '/media/pi/rsync/2023-11-09-pisafe-test.img'
2023-11-09 17:51 ~ Step 3 of 3 - Compressing '/media/pi/rsync/2023-11-09-pisafe-test.img' to '/media/pi/rsync/2023-11-09-pisafe-test.img.zip' ...
2023-11-09 17:51 ~ Running: pv '/media/pi/rsync/2023-11-09-pisafe-test.img' | sudo zip -5 '/media/pi/rsync/2023-11-09-pisafe-test.img.zip' -
2023-11-09 19:05 ~ Backup complete.  -' - 59.5G (/dev/mmcblk0)' backed up to  -'/media/pi/rsync/2023-11-09-pisafe-test.img.zip'  -59.4gb reduced to 35.0gb in 159 min 11 sec.

kPlusPlus, 2023-11-10:
[image]
Result is :) :) :( :(

RichardMidnight, 2023-11-10:
OK, I was confused by the 64GB and the 250GB, I see that you are creating an image of the 250GB. That should work but it will take a lot of time, 4 hours is not unreasonable.  You will reduce the time if you use faster drives... USB3 drives.

I see there is a bug in 1.2.10 that leaves the .img file on the drive.  I am testing now and will release to the beta soon.  You can delete the .img file it left behind.

The backup time for ver 1.2.9 should be the same as 1.2.10.

kPlusPlus, 2023-11-10:
Today im use 1.2.9. Manual copy pisafe from git to this working place. Look for result. Only 201 minutes for all, and image is regular so small. !!!
[image]

kPlusPlus, 2023-11-10:
neofetch -> 
[image]

Disk -> 
[image]
GParted
[image]

Can I help you ,for debug new version ?

RichardMidnight, 2023-11-10:
Interesting... so 1.2.9 works but with 1.2.10 pishrink (step 2) fails.  I will look at it this weekend.  I am glad I left the old one available!!
With 1.2.10 can you run "sudo pisafe".  It that works, then it is most likely a rights issue.

kPlusPlus, 2023-11-12:
> Interesting... so 1.2.9 works but with 1.2.10 pishrink (step 2) fails. I will look at it this weekend. I am glad I left the old one available!! With 1.2.10 can you run "sudo pisafe". It that works, then it is most likely a rights issue.

YeaP with Sudo

RichardMidnight, 2023-11-12:
Thanks.  Just to confirm, "sudo pisafe" with version 1.2.10 does work properly?

closed 2023-11-15 by RichardMidnight