RichardMidnight/pi-safe #44: Tooo loooong

kPlusPlus, 2023-12-16:
Hi again me

Version 1.2 11
for my great wdc old laptop hdd 250Gb over 6 hourse ????? only for 1. step. About cca 16 Mb/s.
Ubuntu 20.04 RPI4 4Gb.
When finish take screenshoot to this.

Can I some speed up this routine ?

kPlusPlus, 2023-12-16:
and crash stop
[image]
[image]

RichardMidnight, 2023-12-16:
The time is because of two major things:  Size of your device, speed of reading and writing devices.

1) Size of your drive.  
  - 250gb is very large.  If it is mostly empty and the last partition is ext4, you can shrink the last partition a bit.  gparted is a popular tool.  Any freespace at the END OF THE DEVICE will be skipped and thus the process will be faster.  For instance, if you shrunk the partition to 128gb, it would take 1/2 the time.  You see where it says Skipping 16.5kb of freespace at end of media.  If there was more freespace at the end of the media, it would skip more and be faster.
 - run "lsblk -f /dev/sda "  and look at the FSUSE%.  That will quickly tell you if you the % of the partition that is in use, and therefore if you can shrink the partition much.
  - I usually work with 32GB drives or smaller.  That being said, it will handle bigger drives, it just takes time as you have noted.
  
2) Speed of your read drive AND write drive.  
 - My pi 4 running RapiOS copies an 8GB card in 9 minutes to a local device.  So if 1GB takes 1.1 minutes.  So 256GB = 281 min = 4.25 hours.  Ubuntu is probably slower than RaspiOS.  Going to a network drive may be slower.  Going across wifi will be slower.

I hope this helps

closed 2023-12-26 by RichardMidnight