RichardMidnight/pi-safe #46: PiSafe WARNING: ~ pv not installed.

BoraFreak, 2024-07-27:
~ $ bash pisafe install
PiSafe 1.2.11

Installing pv...
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
E: Paket pv kann nicht gefunden werden.
PiSafe WARNING: ~ pv not installed.
Checking for PiSafe updates...

RichardMidnight, 2024-07-27:
I only know English and a little Spanish, but I think the error a rights issue.  Try starting the command with su.

BoraFreak, 2024-07-27:
I started pisafe in command line with or without su, but same error "PiSafe WARNING: ~ pv not installed."
it looks that pv package no more available

:~ $ pisafe
:~ $ sudo pisafe
:~ $ bash pisafe install
:~ $ sudo bash pisafe install

always : PiSafe WARNING: ~ pv not installed. Package pv cannot be found.

RichardMidnight, 2024-07-27:
Ok.  It can't find the pv pv program. It uses pv to show progress of activity.  pv is pipe-viewer. Ideally you can find the pv program for your distro and install it then pisafe will see it and use it.  
Alternatively, you can try pisafe without it.  It might work but with little feedback.  I can't remember off-hand. 

What is your hardware and distro?

BoraFreak, 2024-07-27:
I use a Raspberry Pi4 with PiOS 11 Bullseye 32 Bit Version

BoraFreak, 2024-08-24:
Issue is solved, it works again fine with a fresh system.

RichardMidnight, 2024-08-24:
Fantastic.  I hope it is helpful to you.

closed 2024-08-24 by RichardMidnight